daveake/FlexTrack #4: Undesirable Behaviour on loosing lock in GPS.ino

SteveRan, 2016-03-05:
On getting first valid 3D position ProcessUBX_NAV_SOL() sets "HaveHadALock" ProcessUBX_NAV_POSLLH() uses "HaveHadALock" to determine if it should write to "GPS.Longitude" "GPS.Latitude" and "GPS.Altitude" which are then subsequently used in the transmitted position.

If the GPS gets lock - but subsequently looses it (say on payload landing) the transmitted position will be whatever the GPS outputs without lock - which can often be significantly out from the actual position. Under loss of lock it would be better to transmit the last known good position.

Proposal: ProcessUBX_NAV_POSLLH() should qualify "HaveHadALock" with GPS.Lock >= 2 so that there is a least a valid 2d position before updating GPS.Longitude" "GPS.Latitude" and "GPS.Altitude".

daveake, 2016-03-05:
Yeah good point, and I've been meaning to check as I couldn't remember if
I'd accounted for that.  Will get it sorted.

Dave

SteveRan, 2016-03-05:
During ground testing I noticed when it lost lock it was outputting positions a few hundred meters away but at 36Km altitude!
	
	Steve

Steve Randall
Random Engineering Ltd
steve@randomaerospace.com
+44 7802 242135

daveake, 2016-03-05:
If it did the same but 49km altitude I think it would have a customer :p

Dave

daveake, 2020-06-11:
Done.  Now using UBX-NAV-PVT with fixtype 2-4 for lat/lon, and 3-4 for altitude.

closed 2020-06-11 by daveake