daveake/FlexTrack #6: Does lora.ino work with a RFM95W instead of RFM98W

torntrousers, 2016-09-15:
I have a bunch of RFM95's and antennas so Just wondered if you/anyone knows if this lora.ino code would work with an RFM95W on 868Mhz instead of the 434Mhz RFM98W?

Are there any issues with RFM95W  and 868MHz with HAB in the UK I should be aware of?

daveake, 2016-09-15:
Yes it will work fine.

There aren't many 868 receiver stations, and the range is compromised
compared to 434, but it would work.  Make sure you use a
frequency/power/duty cycle that's compliant with IR2030.

Dave

torntrousers, 2016-09-15:
Thanks for the quick reply!

Do you know is the range reduced significantly compared to 434MHz? I was planning a HAB using just LORA for tracking and no other radio transmitter (just to save cost), do you think thats dooable, or fool hardy?

daveake, 2016-09-15:
868 vs 434, everything else equal, has half the range.

You really need to check IR2030 to find out what power/bandwidth/duty cycle
limits there are.  Different parts of the band have different limits.
Typical 434 flights use 10mW and 20.8kHz bandwidth and those will reach the
horizon at 600km or so at altitude, so use that as a guide.

There are probably only 3 868 LoRa HAB receiving stations in the country,
so depending on location, tracking is likely to be just you, so you'll need
to plan the chase and be fairly close by when the flight lands.

434 aerials are a lot easier to come by than 868 ones.

Personally I'd happily fly 868 with no backup, but I've chased a few
flights.  In your case I suggest that a few quid for a 434 module would be
a good investment, and is a small part of the total cost of a flight.

Suggest you ask in the #highaltitude IRC channel on Freenode for further
advice as this isn't specific to Flextrack!

Dave

torntrousers, 2016-09-15:
OK thanks for your comments. I have lurked on #highaltitude IRC so will ask more there.  But trying to sneak in one further question here before closing this...

Hopefully my balloon wont blow too much further than 150Km or so away so line of sight in the sky shouldn't make too much difference between 434MHz and 868MHz. But once landed on the ground I read somewhere (can't find the link now of course) that 868MHz might work better due to the [Fresnel zone](https://en.wikipedia.org/wiki/Fresnel_zone),  and once its on the ground if I can get my receiver up on a hill somewhere nearby I've more chance of hearing the transmitter with 868MHz than with 434MHz due to the Fresnel Zone.

Could there be any truth in that?

daveake, 2016-09-15:
No idea.  Range after landing varies a lot depending on geography; you'd
have to do comparative tests to see if it's a factor.

closed 2016-09-15 by daveake